PyHDI/Pyverilog #12: Slow dataflow analysis

shtaxxx, 2015-07-10:


yushi96, 2019-09-05:
hi，i use dataflow analysis analyze AES design, it had spent one day.now, it still running. i want to know why?

shtaxxx, 2019-09-05:
I think it's because the current dataflow analyzer does not use memorization.

yushi96, 2019-09-05:
Thanks for your reply . so, if I have enough time and LINUX platform has enough memory, no matter how large design always can analyze? such as : c8051

shtaxxx, 2019-09-05:
I think the current analyzer does not handle a realistic size of hardware. I think it should be accelerated, but I don't have enough time for this :(

yushi96, 2019-09-05:
I want to extract dataflow from the verilog design (uart,AES,c8051). But I can't write the verilog compiler front end. so , I read your paper and find pyverilog. about my mission, can you give me some suggestion? thank you very much!

jainpranav1, 2022-11-24:
It would be really helpful if the dataflow analyzer could be sped up.