PyHDI/Pyverilog #16: Enhancement: Support for SystemVeirlog

amal-khailtash, 2015-08-20:
Is there a plan to support SystemVerilog 2012?

shtaxxx, 2015-08-21:
Unfortunately, I don't have any plan to extend the language feature of Pyverilog.

Pyverilog uses Icarus Verilog as a preprocessor. If we don't develop preprocessor by ourselves, Pyverilog overcome the language feature of Icarus Verilog.

I don't know how many features of SystemVerilog Icarus Verilog supports. Do you have any easy pointers for implementing SystemVerilof features with Icarus Verilog?

Thanks

fukatani, 2015-08-21:
One method is convert systemverilog to verilog before preprocessing.↓ http://electronics.stackexchange.com/questions/78834/systemverilog-to-verilog-translation

But part of systemverilog feature are challenging (interface, enum, struct, union).
Personally, my group don't use interface, enum, struct and union. I use only always_comb, always_ff and logic. So I appreciate if Pyverilog support them ( or I will implement. ).

amal-khailtash, 2015-08-21:
As for preprocessor, you can use https://github.com/wsong83/vpreproc that is a standalone port of Wilson Snyder's Verilog-Perl preprocessor.  I can help with the SystemVerilog grammar.  I have worked on PLY version, but it needs some more work.