PyHDI/Pyverilog #32: function integer

shtaxxx, 2017-03-03:
The current parser may not be able to parse the code as below.

```verilog
function integer funcname;
  input invalue;
  // ...
  funcname = 0;
endfunction
```

dmoore375, 2017-04-19:
Can confirm that I just ran into this issue as well. Additionally, the following will fail:
```verilog
function funcname;
  input integer invalue;
  // ...
  funcname = 0;
endfunction
```
 It seems the parser only has a rule for width values in both of these positions, and not for real or integer type declarations here.

sjalloq, 2020-06-02:
This still seems to be an open bug.  What are the work arounds for this?

shtaxxx, 2020-06-02:
I'm now working on the SystemVerilog support of Pyverilog parser. This issue will be resolved in this update.