PyHDI/Pyverilog #36: unsupport concat type when parse dataflow.

hunterzju, 2018-07-16:
Thanks for your excellent work, but I found it unsupport some operation when parse dataflow graph. The error message it returns is : 
```
File "/usr/local/lib/python3.5/dist-packages/pyverilog-1.1.3.dev0-py3.5.egg/pyverilog/dataflow/bindvisitor.py", line 1339, in getDst
    (str(type(left)), str(left)))
pyverilog.utils.verror.FormatError: unsupported AST node type: <class 'pyverilog.vparser.ast.Concat'> <pyverilog.vparser.ast.Concat object at 0x7fa687524f98>
```
it seems that the function `getDst` in `dataflow/bindvisitor.py` does not support vparser.ast.concat operation. 
My verilog source code in attach files. And the top module I want parse is `one_round` module.
[verilogsrc.zip](https://github.com/PyHDI/Pyverilog/files/2196963/verilogsrc.zip)
Hope the tool could be better and I can also do something for the project if possible.

yushi96, 2019-09-05:
你好，请问你也使用pyverilog编译AES设计了吗？我想编译AES，并获得其dataflow，结果跑了好久都没出结果。但我跑uart的时候，很快。请问你知道为什么吗？

hunterzju, 2019-09-05:
> 你好，请问你也使用pyverilog编译AES设计了吗？我想编译AES，并获得其dataflow，结果跑了好久都没出结果。但我跑uart的时候，很快。请问你知道为什么吗？
可能是有解析过程有环，导致无限循环了，貌似pyverilog对有环的情况处理不太好，可以top命令看一下是不是进程一直在吃内存。

yushi96, 2019-09-05:
那你最后跑出来了嘛？我现在需要提取verilog设计中的数据流，我自己写不来编译器，只能找轮子，他这个感觉还可以哦发自我的华为手机-------- 原始邮件 --------主题：Re: [PyHDI/Pyverilog] unsupport concat type when parse dataflow. (#36)发件人：hunterzju 收件人：PyHDI/Pyverilog 抄送：yushi96 <18700192303@163.com>,Comment 
你好，请问你也使用pyverilog编译AES设计了吗？我想编译AES，并获得其dataflow，结果跑了好久都没出结果。但我跑uart的时候，很快。请问你知道为什么吗？
可能是有解析过程有环，导致无限循环了，貌似pyverilog对有环的情况处理不太好，可以top命令看一下是不是进程一直在吃内存。

shtaxxx, 2019-09-05:
Hi, could you use English, please?

hunterzju, 2019-09-05:
> Hi, could you use English, please?

Sorry, we are discussing the problem we met when parse AES module using pyverilog : the memory expand but can not return a result. I guess maybe there is a ring in the module and pyverilog came into a dead loop, but I have no idea how to solve it. 
Sorry again for our behavior.

yushi96, 2019-09-05:
Sorry

klsllzx, 2021-02-05:
I encounter the same situation. The dataflow analyzer cannot parse AES crypto core. Do you guys have any solution?

mostafahosseini72, 2023-06-06:
> Sorry, we are discussing the problem we met when parse AES module using pyverilog : the memory expand but can not return a result. I guess maybe there is a ring in the module and pyverilog came into [...]

> Thanks for your excellent work, but I found it unsupport some operation when parse dataflow graph. The error message it returns is :
>
> ```
> File [...]

hi bro, could you fix the problem if you can tell me by this email hosseinitoa21@gmail.com
thanks a lot

Lan2003, 2025-05-29:
When I use it to parse AES crypto core I encounter the same situation. Do you guys have any solution?