PyHDI/Pyverilog #37: It fails to parse two-dimensional array

chsiensu, 2018-09-06:
Hi,

My p_sram.v and my python file are as attached files and the version of verilog code parser is 1.1.1.
[attached.zip](https://github.com/PyHDI/Pyverilog/files/2355120/attached.zip)
After parsing the p_sram.v, it shows out the below error. I think it parser does not support the two-dimensional array like reg [7:0] test_r [127:0][63:0]. Would you please reproduce my case?
Thank you
-----------error------------------------------------
Generating LALR tables
WARNING: 160 shift/reduce conflicts
Syntax error
Traceback (most recent call last):
  File "./gen_sram.py", line 164, in <module>
    main()
  File "./gen_sram.py", line 116, in main
    preprocess_define=options.define)
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/parser.py", line 2225, in parse
    ast = codeparser.parse()
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/parser.py", line 2211, in parse
    ast = self.parser.parse(text, debug=debug)
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/parser.py", line 64, in parse
    return self.parser.parse(text, lexer=self.lexer, debug=debug)
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/ply/yacc.py", line 265, in parse
    return self.parseopt_notrack(input,lexer,debug,tracking,tokenfunc)
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/ply/yacc.py", line 1047, in parseopt_notrack
    tok = self.errorfunc(errtoken)
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/parser.py", line 2184, in p_error
    self._coord(p.lineno))
  File "/CAD/Anaconda2/4.4.0/lib/python2.7/site-packages/pyverilog/vparser/plyparser.py", line 54, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
pyverilog.vparser.plyparser.ParseError: :22: before: [

RobyB, 2020-03-02:
I have the same issue.
In my case ``` logic  [3:0][3:0] Data``` produces the same error

sjalloq, 2020-06-30:
I'm hitting this with a multi-dimensional parameter of the form:

`localparam [10:0]          CORE_DATA_WIDTH       [NUM_CORES-1:0] = '{256, 256};`

HamzaShabbir517, 2021-01-15:
I want to create an array like
reg [31:0] array [0:15]  
so which pyverilog function should i use as there is no function for array.