PyHDI/Pyverilog #43: PyVerilog does not handle genvar and generate blocks

anandh1791, 2019-03-25:
Hi Anandh,
> 
> Thank you! Now I successfully compiled your code by Icarus verilog.
> I found that some additional features must be implemented in Pyverilog.
> For instance, Pyverilog does not support automatic attribute of function definition.
> 
> Give me a little time to modify it ...
> 
> Thanks,
> 
> Shinya
> 
> On 2019/03/06 1:51, anandh krishnan wrote:
>  > Hello Shinya san,
>  >
>  > I am very sorry, but please add these two lines in your header file:
>  >
>  > parameter P_FLOP_STRUCT_PARITY_ENB    = 1'b0;
>  > parameter integer  P_FLOP_STRUCT_PARITY_GRAN   = 8;
>  >
>  > Arrigato.
>  >
>  > Thank you,
>  > Anandh
>  >
>  > On Monday, March 4, 2019, 8:49:59 PM PST, Shinya TAKAMAEDA-YAMAZAKI <takamaeda@ist.hokudai.ac.jp <mailto:takamaeda@ist.hokudai.ac.jp>> wrote:
>  >
>  >
>  > Hi Anandh,
>  >
>  > Thanks for sharing your codes.
>  >
>  > I still have a complication error on Icarus Verilog.
>  > Can you tell me the definitions of `P_FLOP_STRUCT_PARITY_GRAN and `P_FLOP_STRUCT_PARITY_ENB ?
>  > I could not find the definitions in your main code and header.
>  >
>  > $ iverilog -tnull -Wall tmp.v
>  > tmp.v:29: error: Unable to bind parameter `P_FLOP_STRUCT_PARITY_GRAN' in `ns_s_fifo'
>  > tmp.v:30: error: Unable to bind parameter `P_FLOP_STRUCT_PARITY_GRAN' in `ns_s_fifo'
>  > tmp.v:32: error: Unable to bind parameter `P_FLOP_STRUCT_PARITY_ENB' in `ns_s_fifo'
>  > tmp.v:34: error: Unable to bind parameter `P_FLOP_STRUCT_PARITY_ENB' in `ns_s_fifo'
>  > tmp.v:268: error: Unable to bind parameter `P_FLOP_STRUCT_PARITY_ENB' in `ns_s_fifo'
>  > tmp.v:268: error: Cannot evaluate genvar conditional expression: P_FLOP_STRUCT_PARITY_ENB
>  > tmp.v:268: error: Unable to bind parameter `P_FLOP_STRUCT_PARITY_ENB' in `ns_s_fifo'
>  > tmp.v:268: error: Cannot evaluate genvar conditional expression: P_FLOP_STRUCT_PARITY_ENB
>  > 8 error(s) during elaboration.
>  >
>  > Thanks,
>  >
>  > Shinya
>  >
>  >
>  > On 2019/03/05 13:32, anandh krishnan wrote:
>  >  > Hello Shinya san,
>  >  >
>  >  > Please find the header attached.
>  >  >
>  >  > Thanks,
>  >  > Anandh
>  >  >
>  >  > On Monday, March 4, 2019, 7:03:59 PM PST, Shinya TAKAMAEDA-YAMAZAKI <takamaeda@ist.hokudai.ac.jp <mailto:takamaeda@ist.hokudai.ac.jp> <mailto:takamaeda@ist.hokudai.ac.jp <mailto:takamaeda@ist.hokudai.ac.jp>>> wrote:
>  >  >
>  >  >
>  >  > Hi Anandh,
>  >  >
>  >  > I'm sorry for the very late response, again.
>  >  >
>  >  > I tried to read your code by Veriloggen, but I could not do that because I don't have "ns_common_func.vh".
>  >  > Can you share your header file?
>  >  >
[... 595 lines cut ...]
>  >  >  >  >  >     end
>  >  >  >  >  > endgenerate
>  >  >  >  >  >
>  >  >  >  >  > wire [P_DATA_GROUP_WIDTH-1:0] data_ppln_exit;
>  >  >  >  >  > wire [P_DATA_GROUP_WIDTH-1:0] data_ppln_exit_clean;
>  >  >  >  >  >
>  >  >  >  >  > assign data_ppln_exit =  data_ppln[P_NUM_OF_STAGES];
>  >  >  >  >  >
>  >  >  >  >  > `ifdef NS_ECC_X_SQUASH_DEGREE2
>  >  >  >  >  >     genvar x;
>  >  >  >  >  >     generate
>  >  >  >  >  >     endgenerate
>  >  >  >  >  >     for (x=0; x<P_DATA_GROUP_WIDTH; x=x+1) begin : G_DATA
>  >  >  >  >  >        assign data_ppln_exit_clean[x] = (data_ppln_exit[x] === 1'bx) ? in_data_bundle[x]: data_ppln_exit[x];
>  >  >  >  >  >     end
>  >  >  >  >  > `else
>  >  >  >  >  >     assign data_ppln_exit_clean = data_ppln_exit;
>  >  >  >  >  > `endif
>  >  >  >  >  >
>  >  >  >  >  > assign {o_flit_data, o_flit_sb, o_flit_usrsb, o_flit_type, o_flit_mcpkt, o_flit_sop, o_flit_eop, o_flit_bv, o_flit_outp,
>  >  >  >  >  >          o_pktctl_parity, o_rinfo_parity, o_data_parity, o_usrsb_parity} =  data_ppln_exit_clean;
>  >  >  >  >  > assign o_flit_valid = valid_ppln[P_NUM_OF_STAGES] ;
>  >  >  >  >  > assign o_cg_busy = busy_ppln[P_NUM_OF_STAGES] ;
>  >  >  >  >  >
>  >  >  >  >  > endmodule

shtaxxx, 2019-03-29:
main.v

```verilog
module ns_s_fifo(
   // Outputs
   rd_data,
   rd_data_valid,
   occupancy_cnt,
   fifo_full,
   perr,
   // Inputs
   clk,
   clk_g,
   reset,
   wr_enb,
   wr_data,
   wr_parity,
   rd_enb,
   perr_inj
   );

`include "ns_common_func.vh"

   localparam P_FPGA_INFER_RAM        = 1'b0;
   parameter P_FIFO_D_WIDTH           = 32;
   parameter P_FIFO_DEPTH             = 4;
   parameter P_REG_OUT                = 1'b0;
   parameter P_OCCUPANCY_CNT          = 1'b0;

   parameter P_EXT_PARITY_GEN = 1'b1;
   localparam P_DATA_WIDTH_REMAINDER   = P_FIFO_D_WIDTH % P_FLOP_STRUCT_PARITY_GRAN ;
   localparam P_DATA_PAD_BITS          = (P_DATA_WIDTH_REMAINDER==0) ? 0 : P_FLOP_STRUCT_PARITY_GRAN - P_DATA_WIDTH_REMAINDER;
   localparam P_DATA_PLUS_PAD_WIDTH    = P_FIFO_D_WIDTH + P_DATA_PAD_BITS ;
   localparam P_EXT_PARITY_BITS        = ((P_EXT_PARITY_GEN == 1'b0) | (P_FLOP_STRUCT_PARITY_ENB == 1'b0)) ? 1 :
                                          P_DATA_PLUS_PAD_WIDTH / P_FLOP_STRUCT_PARITY_GRAN ;
   localparam P_NUM_PARITY_BITS        = P_FLOP_STRUCT_PARITY_ENB ? P_DATA_PLUS_PAD_WIDTH / P_FLOP_STRUCT_PARITY_GRAN : 1 ;

   

   localparam LP_ONE = 1;
   
   localparam P_FIFO_DEPTH_MOD = ((P_FIFO_DEPTH <= 2) & P_REG_OUT) ? 2 :
                                 ((P_FIFO_DEPTH >= 3) & P_REG_OUT) ?    (P_FIFO_DEPTH-1) : 
                                  (P_FIFO_DEPTH <= 1)              ? 2 : P_FIFO_DEPTH ;

   localparam P_FIFO_A_WIDTH = (P_FIFO_DEPTH_MOD == 1) ? 1 : clogb2(P_FIFO_DEPTH_MOD) ;
   localparam P_OCCU_CNTR_WIDTH = clog2(P_FIFO_DEPTH+1);
   
   localparam [P_FIFO_A_WIDTH-1:0] P_FIFO_DEPTH_MIN_1 = P_FIFO_DEPTH_MOD-1 ;

   input                          clk;
   input                          clk_g;
   input                          reset;

   input                          wr_enb;
   input [P_FIFO_D_WIDTH-1:0]     wr_data;
   input [P_EXT_PARITY_BITS-1:0]  wr_parity;

   input                          rd_enb;
   output [P_FIFO_D_WIDTH-1:0]    rd_data;
   output                         rd_data_valid;

   output [P_OCCU_CNTR_WIDTH-1:0] occupancy_cnt;
   
   output                         fifo_full;
  
   // NS_C: Flop Structure Parity I/O
   // lint_checking USEPRT off   
   input                          perr_inj;
   // lint_checking USEPRT on
[... 398 lines cut ...]
endfunction

// NOTE: beware of unsigned/signed conversion
function automatic integer min_int;
   input integer a;
   input integer b;
   begin
      min_int = a<=b ? a : b;
   end
endfunction

function automatic [5:0] nibble_count;
   input [31:0] value;
   begin
      if ((value % 4) == 0)
          nibble_count = value/4 ;
      else
          nibble_count = (value/4) + 1 ;
   end
endfunction

// pragma coverage on
//lint_checking USEFTN ON

```

shtaxxx, 2019-03-29:
```bash
python3 example_parser.py ~/work/tmp/ns_common.v -I ~/work/tmp/
```

results

```
Syntax error
Traceback (most recent call last):
  File "example_parser.py", line 52, in <module>
    main()
  File "example_parser.py", line 45, in main
    preprocess_define=options.define)
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/parser.py", line 2248, in parse
    ast = codeparser.parse()
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/parser.py", line 2234, in parse
    ast = self.parser.parse(text, debug=debug)
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/parser.py", line 64, in parse
    return self.parser.parse(text, lexer=self.lexer, debug=debug)
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/ply/yacc.py", line 265, in parse
    return self.parseopt_notrack(input,lexer,debug,tracking,tokenfunc)
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/ply/yacc.py", line 1047, in parseopt_notrack
    tok = self.errorfunc(errtoken)
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/parser.py", line 2207, in p_error
    self._coord(p.lineno))
  File "/Users/shinya/work/pyverilog/pyverilog/vparser/plyparser.py", line 54, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
pyverilog.vparser.plyparser.ParseError: :41: before: integer
```

shtaxxx, 2019-03-29:
At first, pyverilog does not support "automatic".

shtaxxx, 2019-03-29:
I made another issue for "automatic" support.