PyHDI/Pyverilog #56: test error

sheriefreda, 2020-01-23:
Hi, I followed installation instructions on MacOS 10.14.6 and all is good. Then I get this error with the example_parser on test.v

Traceback (most recent call last):
  File "examples/example_parser.py", line 10, in <module>
    import pyverilog.utils.version
ModuleNotFoundError: No module named 'pyverilog.utils.version'

ray-dev1, 2020-02-21:
I just encountered this exact error on Linux. 
Were you able to resolve it? 

Traceback (most recent call last):
  File "pyverilog/examples/example_parser.py", line 10, in <module>
    import pyverilog.utils.version
ModuleNotFoundError: No module named 'pyverilog.utils.version'

pawanpatil94, 2020-04-16:
You can **import pyverilog** and replace the line
VERSION = pyverilog.utils.version.VERSION
with
VERSION = pyverilog.__version__