PyHDI/Pyverilog #58: How to debug a yacc parse error?

sjalloq, 2020-06-02:
Hi,

I'm hitting a parse error that seems to come from YACC but it's no giving a very useful error message.  I don't even know what file it is parsing as I'm traversing a hierarchy.

Is there a way to get more info that just the following?

pyverilog.vparser.plyparser.ParseError: :735: before: (

shtaxxx, 2020-06-02:
VerilogParser accept "debug" option to enable the debug mode.
```
        self.parser = VerilogParser(outputdir=outputdir, debug=debug)
```

To obtain an actual debugging log, the debug option in the parse method must be also enabled.
```
        ast = self.parser.parse(text, debug=debug)
```

In the develop branch, enabling debug option at the top-level 'parse' method, you can get the debugging log. Please see "examples/example_parser.py" .

```
    ast, directives = parse(filelist,
                            preprocess_include=options.include,
                            preprocess_define=options.define, debug=True)
```

shtaxxx, 2020-06-02:
PLY provides some information of the parser construction. parser.out shows the parsing rules. I think the parsing logs using "debug" mode and this parsing rule output will be useful for the debugging.

sjalloq, 2020-06-30:
Something like the following would be useful in order to match the line number in the error to the pre-processed Verilog:

```
2302,2307c2300
<         try:
<             ast = self.parser.parse(text, debug=debug)
<         except Exception as e:
<             with open('preproc.v', 'w') as f:
<                 f.write(text)
<             raise e
---
>         ast = self.parser.parse(text, debug=debug)
```