PyHDI/Pyverilog #77: function in module can not be parse, and get error

asky26, 2021-01-10:
Test the  module which contain the funtion code in it, then I got error.
I check the code of lex. It has the key word "function". So I guess that I have to modify yacc to make it work well.

I already have run the pyverilog well with some simple verilog code which do not have function in it.

shtaxxx, 2021-01-10:
Which version?

asky26, 2021-01-10:
1.2.1
I use function to cal the port width.

shtaxxx, 2021-01-10:
Can you show your code?

asky26, 2021-01-10:
`module top_module#(parameter in_w = 1)( input in, output out );
    parameter out_w = do_math(in_w);
    wire [out_w-1 : 0] out;
	function output_width;
		input input_width; 
        begin
	 	do_math = input_width + 1;
        end
	endfunction
	assign out = in + 1;
endmodule`

I do not have the code rigth now. It is hard to copy code from my computer for the work.
So I write the simple code.

shtaxxx, 2021-01-10:
```
module top_module#(parameter in_w = 1)( input in, output out );
parameter out_w = do_math(in_w);
wire [out_w-1 : 0] out;
function output_width;
input input_width;
begin
do_math = input_width + 1;
end
endfunction
assign out = in + 1;
endmodule
```

Before testing your code on Pyverilog 1.2.1, I passed your code to Icarus Verilog, and it says the code has an error.

```
iverilog -tnull -Wall test.v
```

```
test.v:2: error: No function named `do_math' found in this context (top_module).
1 error(s) during elaboration.
```

asky26, 2021-01-11:
Sorry for my mistake. Here is the code that I have run with the iverilog.
//////////////
module top_module#(parameter in_w = 1)( 
	input in, 
	output [out_w-1 : 0] out );
	localparam out_w = output_width(in_w);
	
	
	function integer output_width;
		input input_width;
	begin
		output_width = input_width + 1;
	end
endfunction
	assign out = in + 1;
endmodule

asky26, 2021-01-21:
I have test the code with version 1.3.0. But I got this:
pyverilog.utils.verror.FormatError: unsupported AST node type: <class 'pyverilog.vparser.ast.FunctionCall'> output_width

Can you fix it? or is there a way to skip this error.

shtaxxx, 2021-01-21:
```bash
 python3 example_parser.py ~/tmp/yourcode.v
```

```
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Source:  (at 1)
  Description:  (at 1)
    ModuleDef: top_module (at 1)
      Paramlist:  (at 1)
        Decl:  (at 1)
          Parameter: in_w, False (at 1)
            Rvalue:  (at 1)
              IntConst: 1 (at 1)
      Portlist:  (at 1)
        Ioport:  (at 2)
          Input: in, False (at 2)
        Ioport:  (at 3)
          Output: out, False (at 3)
            Width:  (at 3)
              Minus:  (at 3)
                Identifier: out_w (at 3)
                IntConst: 1 (at 3)
              IntConst: 0 (at 3)
      Decl:  (at 4)
        Localparam: out_w, False (at 4)
          Rvalue:  (at 4)
            FunctionCall:  (at 4)
              Identifier: output_width (at 4)
              Identifier: in_w (at 4)
      Function: output_width (at 6)
        Width:  (at 6)
          IntConst: 31 (at 6)
          IntConst: 0 (at 6)
        Decl:  (at 7)
          Input: input_width, False (at 7)
        Block: None (at 8)
          BlockingSubstitution:  (at 9)
            Lvalue:  (at 9)
              Identifier: output_width (at 9)
            Rvalue:  (at 9)
              Plus:  (at 9)
                Identifier: input_width (at 9)
                IntConst: 1 (at 9)
      Assign:  (at 12)
        Lvalue:  (at 12)
          Identifier: out (at 12)
        Rvalue:  (at 12)
          Plus:  (at 12)
            Identifier: in (at 12)
            IntConst: 1 (at 12)
```

There is no error in parsing.

As you pointed out, dataflow analyzer might have no rules for `function integer funcname`.

Do you need the dataflow analyzer? I am very appreciated if you suggest how to modify the dataflow analyzer.

asky26, 2021-01-27:
Maybe function can be processed in 2 ways.
1) Fcuntion is called in module, just as the example. The best way should analyize the function code, and get the return value of th function called. But it seems like a hard way.  In actually, function code is not my key point, but some modules in filelist have function in themself, so the dataflow-analyzer can not keey going.
Maybe set some options for me, to make dataflow analyzer skip the function. For example, leave the the code that the function is  called. The most verilog-code do not have the function. 
I want to use dataflow analyzer to make some rule check. 
2) Fucntion is used for simulation.  It can be skiped, I guess

Thank you for your help.

asky26, 2021-05-04:
I have make some local modification. And I got this:
[image]
It works, with some wrong infomation. 
I am not totally understand source code.  Can I get some deltailed instructions?

INSIinc, 2021-09-27:
function in module can not be parse, and get error
[image]

How do I handle this error?

vivoo94, 2022-10-02:
Test the module which contain the funtion code Parser in it, then I got error. and it is not parsing I am using version 1.3 

  File "C:\Users\alfredMoussa\Anaconda3\lib\site-packages\pyverilog\vparser\parser.py", line 2340, in parse
    ast = codeparser.parse()

  File "C:\Users\alfredMoussa\Anaconda3\lib\site-packages\pyverilog\vparser\parser.py", line 2317, in parse
    text = self.preprocess()

  File "C:\Users\alfredMoussa\Anaconda3\lib\site-packages\pyverilog\vparser\parser.py", line 2310, in preprocess
    text = open(self.preprocess_output).read()

FileNotFoundError: [Errno 2] No such file or directory: 'preprocess.output'