PyHDI/Pyverilog #81: Can it work on hierarchy of design?

jadonrahul, 2021-03-10:
1) I want to run  parser on multiple files. instead of one by one can it accept filelist?
2) Can it follow the hierarchy of design?

vrrao01, 2022-11-17:
@jadonrahul  Did you find an answer to this? I have to use Pyverilog for a similar use case and was wondering if parsing multiple files is possible easily. Or do we need to keep track of the instantiations across files.

mostafahosseini72, 2022-11-27:
> 1. I want to run  parser on multiple files. instead of one by one can it accept filelist?
> 2. Can it follow the hierarchy of design?

hi bro, could you find the answer, if you could please send my the solution by email.
thank you.
my email : mostafa.hosseini@znu.ac.ir