PyHDI/Pyverilog #84: ast.IntConst creats collisions on dicts

Lucaz97, 2021-03-16:
The parent class ast,Node defines the __eq__ and __hash__ methods in a way that with two ast.IntConst objects with the same value a collision occurs when using such objects as keys in a dict.

Such methods should guarantee that such thing does not occur. I fixed the problem locally by changing the following line 
https://github.com/PyHDI/Pyverilog/blob/5847539a9d4178a521afe66dbe2b1a1cf36304f3/pyverilog/vparser/ast.py#L246
with 
`attr_names = ('value','lineno')`

If you'd like I can do a pull request.

shtaxxx, 2021-04-26:
Thanks for the nice suggestion.
Could you show us what kind of adverse effects are occurred by the current implementation?
Furthermore, I wonder just adding "lineno" is sufficient.

Lucaz97, 2021-04-26:
@shtaxxx Thank you for your reply.

Let's suppose I am parsing this FIR filter implementation: 
https://github.com/mit-ll/CEP/blob/master/hdl_cores/dsp/FIR_filter.v
I have a dictionary that uses as a key some ast nodes, some of which are ast.Constant.
In the design above there are some constants of the same values (for example line 74 and 85 both have a constant of value 3).
With the current implementation I get collisions on the dict and I only detect the last constant for each value present in the design (I lose the 3 at line 74).  
Adding lineno to the attr allows for differentiating those cases.

It is true that if I have a line like:

something = c1 + c2 +c3 +c4 .... 

with c1 ... cN constants of the same values, they would still collide. 
Apparently I did not have any since I only thought about it now and my tests are passing.