PyHDI/Pyverilog #109: dataflow analysis error

EDA-Testing, 2022-08-08:
Hello, I am using Pyvrilog's dataflow analyzer. But it is very strange that he saw an error in the process. The error does not seem to be due to my code, but to the fact that there is no module named Insertcond in Pyverilog. The specific error message is as follows:
**### AttributeError: module 'pyverilog.dataflow.reorder' has no attribute 'insertCond'**
Do you have any solution to help me solve these problems?

EDA-Testing, 2022-08-08:
I checked the reoder file and the file that reported the error. Only the InsertCond module is missing, I don't know if this is a bug

Wenz-jam, 2024-07-19:
I found that in example_graphgen.py at line 101, the example code uses reorder=options.reorder, but the parameter name changed to do_reorder in commit 7086786a6f5f8ad965b097344a77b4b86d21a93c. After I tried changing reorder to do_reorder in the example code, the example code ran successfully. Although this doesn't actually solve the issue of the missing function, it seems that doing so prevents the function missing error from occurring.

``` diff
99    for target in options.searchtarget:
100        graphgen.generate(target, walk=options.walk, identical=options.identical,
---  101                        step=options.step, reorder=options.reorder, delay=options.delay)
+++  101                        step=options.step, do_reorder=options.reorder, delay=options.delay)

```

ZhiyuanYan, 2024-10-15:
> Hello, I am using Pyvrilog's dataflow analyzer. But it is very strange that he saw an error in the process. The error does not seem to be due to my code, but to the fact that there is no module [...]

Hi, I also got the same error. Have you solved this problem?

cemery123, 2024-10-15:
> Hi, I also got the same error. Have you solved this problem?

Nope, it seems like there are some errors in PyVerilog and i do not have enough time to revise it. So i choose yosys to analyse my Verilog code, `Yosys show`instruction. For controlflow detection i suggest you can try Vivado or Quaturs. But if you have enough time it will be possilble for you to modify the oral code of Pyverilog. I do not know if this reply can help you, but i hope it can find you well.

ZhiyuanYan, 2024-10-15:
> Nope, it seems like there are some errors in PyVerilog and i do not have enough time to revise it. So i choose yosys to analyse my Verilog code, `Yosys show`instruction. For controlflow detection i [...]

Receive it. It can help me a lot. Thank you very much!

cemery123, 2024-10-15:
> Receive it. It can help me a lot. Thank you very much!

My pleasure, Good luck in your Phd journey.

zsxpdsyz, 2025-07-12:
Hello, @cemery123 @ZhiyuanYan 

I think I have solved this issue, it may be caused by the author using an incorrect function call during programming.

You can fix this issue by changing the function call in the source code at the following position:
https://github.com/PyHDI/Pyverilog/blob/81838bc463d17148ef6872af34eb27585ee349ba/pyverilog/dataflow/bindvisitor.py#L1031

to `return reorder.insertBranch(cond_df, true_df, false_df);`.