PyHDI/Pyverilog #116: Unsized numbers without base not supported

Louis-DR, 2023-02-19:
Hello,

I noticed that unsized literal number constants with the base specified (`'d0`) are supported but if the base is not specified (`'d`) it raises a ParseError exception :

``` verilog
module top (
  output [7:0] q
);

assign q = '0;

endmodule
```
``` bash
>>> python3 examples/example_parser.py top.v
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Traceback (most recent call last):
  File "/Users/louis/Workspace/Pyverilog/examples/example_parser.py", line 55, in <module>
    main()
  File "/Users/louis/Workspace/Pyverilog/examples/example_parser.py", line 45, in main
    ast, directives = parse(filelist,
                      ^^^^^^^^^^^^^^^
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/parser.py", line 2371, in parse
    ast = codeparser.parse()
          ^^^^^^^^^^^^^^^^^^
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/parser.py", line 2349, in parse
    ast = self.parser.parse(text, debug=debug)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/parser.py", line 77, in parse
    return self.parser.parse(text, lexer=self.lexer, debug=debug)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/ply/yacc.py", line 1063, in parseopt_notrack
    lookahead = get_token()     # Get the next token
                ^^^^^^^^^^^
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/lexer.py", line 51, in token
    return self.lexer.token()
           ^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/ply/lex.py", line 386, in token
    newtok = self.lexerrorf(tok)
             ^^^^^^^^^^^^^^^^^^^
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/lexer.py", line 259, in t_error
    self._error(msg, t)
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/lexer.py", line 263, in _error
    self.error_func(msg, location[0], location[1])
  File "/Users/louis/Workspace/Pyverilog/pyverilog/vparser/parser.py", line 67, in _lexer_error_func
    raise ParseError('%s: %s' % (coord, msg))
pyverilog.vparser.parser.ParseError:  line:5 column:13: Illegal character "'"
```

Note that some tools appear to treat `'0` and `'d0` differently : [ModelSim](https://stackoverflow.com/questions/66665910/why-doesnt-d0-extend-the-full-width-of-the-signal-as-0-does)

I have never worked with lex or yacc but I might dig into it to try implementing this.

YSJL, 2023-10-26:
Hey Louis were you able to fix this implementation?

YSJL, 2023-10-26:
For whoever is going through this issue, you can edit lexer.py, parser.py, ast.py using already existing structures.

For `lexer,py`:
- Add at line 89 within `tokens = keywords + operators + ()`
`'UNSIZEDNUMBER',`
- line 184
  `unsized_number = '\'[0-1xXzZ?]'`
- line 218
  `@TOKEN(unsized_number)`  
  `def t_UNSIZEDNUMBER(self, t):`  
        `   return t`

For  `parser.py`:
- line 1266
  `def p_const_expression_unsizednum(self, p):`  
        `  'const_expression : unsizednumber'`
        `  p[0] = UnsizedConst(p[1], lineno=p.lineno(1))`
        `  p.set_lineno(0, p.lineno(1))`
- line 1281
  ` def p_unsizednumber(self, p):`
        `  'unsizednumber : UNSIZEDNUMBER'`
        `  p[0] = p[1]`
        `  p.set_lineno(0, p.lineno(1))`
- line 1498
  `def p_delays_unsizednumber(self, p):`  
        `  'delays : DELAY unsizednumber'`  
        `  p[0] = DelayStatement(UnsizedConst(`  
        `    p[2], lineno=p.lineno(1)), lineno=p.lineno(1))`
        `  p.set_lineno(0, p.lineno(1))`

For `ast.py`:
- line 267
`class UnsizedConst(Constant):`  
`  pass`