PyHDI/Pyverilog #124: Parse Error: Cannot parse a function

KatCe, 2023-10-15:
Hello, the following module can not be parsed:

```
module minimized_in (
	clk_i,
	rst_ni,
	use_sign_i,
	sign_i,
	result_o
);
	input wire clk_i;
	input wire rst_ni;
	input wire use_sign_i;
	input wire sign_i;
	output wire [63:0] result_o;
	function automatic [2:0] sv2v_cast_19B92;
		input reg [2:0] inp;
		sv2v_cast_19B92 = inp;
	endfunction
endmodule
```

Got the following error:

```
raise ParseError("%s: %s" % (coord, msg))
pyverilog.vparser.parser.ParseError:  line:13: before: "["
```

Pyverilog commit: 81838bc463d17148ef6872af34eb27585ee349ba

----

My setup:
Python 3.9.5
iverilog/focal,now 10.3-1build1 amd64 [installed]
Name: Jinja2
Version: 3.1.2
Name: ply
Version: 3.11

RasheedKibria, 2023-11-03:
Hi Katharina, 

Please use this following Verilog code supported by Pyverilog:

module minimized_in (
	clk_i,
	rst_ni,
	use_sign_i,
	sign_i,
	result_o
);
	input wire clk_i;
	input wire rst_ni;
	input wire use_sign_i;
	input wire sign_i;
	output wire [63:0] result_o;

	function [2:0] sv2v_cast_19B92;
		input [2:0] inp;
		sv2v_cast_19B92 = inp;
	endfunction

endmodule

Thanks,
Rasheed Kibria

KatCe, 2023-11-05:
Hello Rasheed,
The above example is valid according to IEEE standard 1800-2017, "A.2.6 Function declarations". The code was generated by the sv2v (https://github.com/zachjs/sv2v) tool. 
Best regards,
Katharina